FreeJoy-Team/FreeJoy #88: disconnection with rs232 pedals mod

pipes80-collab, 2021-01-25:
after creating a cable as shown in the video, that is a female rs232 that could be connected to Arduino and also to the Blue pill, I also think of the stm32f103, it requires two power sources, one 3.3 and the other 5 times, it only works by connecting to a  PIN, whether 3.3 or 5 volts, nothing changes but trying with both the device becomes an unrecognized usb in the device manager and is no longer detected by the freejoy app.
Here the video https://youtu.be/k5CtXG1rpZg

vostrenkov, 2021-01-25:
You should use 3.3V supply for your potentiometers for proper work. 
If your device gets unrecognised after connecting the connector it should be a wiring problem. Check that you don't have shorts between power and ground

pipes80-collab, 2021-01-25:
The problem is only to connect 3.3 volt and 5 volt like the video

pipes80-collab, 2021-01-25:
Only one 3.3 supply can for 3 potentiometer?

vostrenkov, 2021-01-25:
You don't need to connect 5V to 3.3V. You should use 3.3V instead of 5V. So all red wires should be connected to 3.3V.
STM32 use 3.3V voltage level for everything, and USB is the only part where 5V is used

vostrenkov, 2021-01-25:
> Only one 3.3 supply can for 3 potentiometer?

Yes, common 3.3V, common GND, and individual signal wires

closed 2021-01-25 by pipes80-collab