FreeJoy-Team/FreeJoy #90: G27 encoder

LosTigeros, 2021-01-26:
Hello.

Is it possible to use bultin encoder in G27?
Processor in the motherboard of my G27 died so I'm looking for alternatives.

vostrenkov, 2021-01-26:
Hi,
yes, it is possible to use it as "fast encoder" on A8 and A9 pins

LosTigeros, 2021-01-26:
Okay, thanks for that great project!

LosTigeros, 2021-01-27:
How can I set it up as X axis? After connecting it to pins A8 as A and A9 as B and setting Axis source as Encoder it just switches between -1 and 0 when the wheel is rotating.
[image]

vostrenkov, 2021-01-27:
Did you set A8 and A9 to Fast_Encoder on Pins tab?

vostrenkov, 2021-01-27:
Also your wiring should include pull-up resistors for the encoders inputs
https://www.github.com/FreeJoy-Team/FreeJoyWiki/tree/master/images%2FA1.7.jpg

LosTigeros, 2021-01-27:
Pins:
[image]

Connection:
[image]

LosTigeros, 2021-01-28:
When rotating the encoder I get:
`80mV` or `4.37V` on `A8`
`80mV` or `1.137V` on `A9`

closed 2021-01-28 by LosTigeros