FreeJoy-Team/FreeJoy #91: Error "`FLASH' overflowed" during build the project

armor66, 2021-01-28:
I try to build the project, but even with no customizing main.h get error

FreeJoy-master/armgcc $ make
/usr/libexec/gcc/arm-none-eabi/ld: build/boot/Bootloader.elf section `.data' will not fit in region `FLASH'
/usr/libexec/gcc/arm-none-eabi/ld: region `FLASH' overflowed by 204 bytes
collect2: error: ld returned 1 exit status

FreeJoy-1.6.2b4/armgcc $ make
/usr/libexec/gcc/arm-none-eabi/ld: build/boot/Bootloader.elf section `.text' will not fit in region `FLASH'
/usr/libexec/gcc/arm-none-eabi/ld: region `FLASH' overflowed by 876 bytes
collect2: error: ld returned 1 exit status

$gcc-config -l
 [1] arm-none-eabi-9.3.0
 [2] arm-none-eabi-10.2.0 *
$arm-none-eabi-ld --version
GNU ld (Gentoo 2.34 p6) 2.34.0

wizhippo, 2021-01-28:
You could try changing the optimisation to size to get the few needed bytes.

OPT = -Os

armor66, 2021-01-28:
Thank you, didn't occur to me

closed 2021-01-28 by armor66