FreeJoy-Team/FreeJoy #94: Clean checkout of tag v1.6.2b4 - not building a working fw

flindt, 2021-02-04:
I am running FreeJoy on a STM32-Nucleo board. The released v1.6.2b4 build is working, and connects as a joystick when plugged in.

I have been trying to build it from source, but then the device fails to enumerate. Both master and the tag v1.6.2b4 behaves the same.
The resulting build is considerably smaller (~48kB) than the released (exactly 64kB).

Perhaps I have missed a build option?

flindt, 2021-02-05:
Version v1.5.1b7 works for me as well when I build it from source.

Should I just branch from here for now?

wizhippo, 2021-02-05:
I just submitted a PR https://github.com/FreeJoy-Team/FreeJoy/pull/95

The firmware has 2 parts now.  When you build you will have 2 files.

buil/boot/Bootloader.bin
build/app/FreeJoy.bin

First flash buil/boot/Bootloader.bin using your tool of choice.

Then reboot and hold boot1 high and the device will go into bootloader mode.

Then within the QT configurator it will detect it as a flasher.  Select the FreeJoy.bin file.

Make sure to set boot1 back to low.

flindt, 2021-02-06:
Ok thanks for the quick answer. 

I am not able to run the configurator since I am on a linux only system. 
(are there any other linux users for FreeJoy?)

Was able to make it work by writing both bootloader and freejoy with openocd.

Reksotiv, 2021-02-06:
https://github.com/FreeJoy-Team/FreeJoyConfiguratorQt supports linux.

flindt, 2021-02-06:
Do you happen to know how? The appimage in the release folder segfaults with no further explanation.
I have tried it on two different pc's (both debian 10).

flindt, 2021-02-06:
strace Downloads/FreeJoyQt-x86_64.AppImage 
execve("Downloads/FreeJoyQt-x86_64.AppImage", ["Downloads/FreeJoyQt-x86_64.AppIm"...], 0x7ffcc33cff60 /* 38 vars */) = 0
brk(NULL)                               = 0x18d8000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=225187, ...}) = 0
mmap(NULL, 225187, PROT_READ, MAP_PRIVATE, 3, 0) = 0x7f12e8472000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0000\21\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=14592, ...}) = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f12e8470000
mmap(NULL, 16656, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f12e846b000
mmap(0x7f12e846c000, 4096, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1000) = 0x7f12e846c000
mmap(0x7f12e846d000, 4096, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2000) = 0x7f12e846d000
mmap(0x7f12e846e000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2000) = 0x7f12e846e000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libpthread.so.0", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0@l\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=146968, ...}) = 0
mmap(NULL, 132288, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f12e844a000
mmap(0x7f12e8450000, 61440, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x6000) = 0x7f12e8450000
mmap(0x7f12e845f000, 24576, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x15000) = 0x7f12e845f000
mmap(0x7f12e8465000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1a000) = 0x7f12e8465000
mmap(0x7f12e8467000, 13504, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f12e8467000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libz.so.1", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\0\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\320#\0\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0644, st_size=121280, ...}) = 0
mmap(NULL, 2216336, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f12e822c000
mprotect(0x7f12e8248000, 2097152, PROT_NONE) = 0
mmap(0x7f12e8448000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1c000) = 0x7f12e8448000
close(3)                                = 0
openat(AT_FDCWD, "/lib/x86_64-linux-gnu/libc.so.6", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\2\1\1\3\0\0\0\0\0\0\0\0\3\0>\0\1\0\0\0\260A\2\0\0\0\0\0"..., 832) = 832
fstat(3, {st_mode=S_IFREG|0755, st_size=1824496, ...}) = 0
mmap(NULL, 1837056, PROT_READ, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0x7f12e806b000
mprotect(0x7f12e808d000, 1658880, PROT_NONE) = 0
mmap(0x7f12e808d000, 1343488, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x22000) = 0x7f12e808d000
mmap(0x7f12e81d5000, 311296, PROT_READ, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x16a000) = 0x7f12e81d5000
mmap(0x7f12e8222000, 24576, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x1b6000) = 0x7f12e8222000
mmap(0x7f12e8228000, 14336, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0x7f12e8228000
close(3)                                = 0
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f12e8069000
arch_prctl(ARCH_SET_FS, 0x7f12e8069b80) = 0
mprotect(0x7f12e8222000, 16384, PROT_READ) = 0
mprotect(0x7f12e8448000, 4096, PROT_READ) = 0
mprotect(0x7f12e8465000, 4096, PROT_READ) = 0
mprotect(0x7f12e846e000, 4096, PROT_READ) = 0
mprotect(0x7f12e84d0000, 4096, PROT_READ) = 0
munmap(0x7f12e8472000, 225187)          = 0
set_tid_address(0x7f12e8069e50)         = 15391
set_robust_list(0x7f12e8069e60, 24)     = 0
rt_sigaction(SIGRTMIN, {sa_handler=0x7f12e84506b0, sa_mask=[], sa_flags=SA_RESTORER|SA_SIGINFO, sa_restorer=0x7f12e845c730}, NULL, 8) = 0
rt_sigaction(SIGRT_1, {sa_handler=0x7f12e8450740, sa_mask=[], sa_flags=SA_RESTORER|SA_RESTART|SA_SIGINFO, sa_restorer=0x7f12e845c730}, NULL, 8) = 0
rt_sigprocmask(SIG_UNBLOCK, [RTMIN RT_1], NULL, 8) = 0
prlimit64(0, RLIMIT_STACK, NULL, {rlim_cur=8192*1024, rlim_max=RLIM64_INFINITY}) = 0
brk(NULL)                               = 0x18d8000
brk(0x18f9000)                          = 0x18f9000
openat(AT_FDCWD, "/proc/self/exe", O_RDONLY) = 3
fstat(3, {st_mode=S_IFREG|0755, st_size=29372456, ...}) = 0
read(3, "\177ELF\2\1\1\0AI\2\0\0\0\0\0\2\0>\0\1\0\0\0\220F@\0\0\0\0\0"..., 4096) = 4096
lseek(3, 0, SEEK_SET)                   = 0
read(3, "\177ELF\2\1\1\0AI\2\0\0\0\0\0\2\0>\0\1\0\0\0\220F@\0\0\0\0\0"..., 4096) = 4096
lseek(3, 184320, SEEK_SET)              = 184320
read(3, "\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0"..., 4096) = 4096
read(3, "\24\327\2\0\0\0\0\0\17\1\0\0\0\0\0\0\0\0\0\0\0\0\0\0\1\0\0\0\0\0\0\0"..., 4096) = 4096
close(3)                                = 0
readlink("/proc/self/exe", "/home/flindt/Downloads/FreeJoyQt"..., 4096) = 48
futex(0x7f12e846f0c8, FUTEX_WAKE_PRIVATE, 2147483647) = 0
[... 69 lines cut ...]
rt_sigaction(SIGINT, NULL, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=0}, 8) = 0
rt_sigaction(SIGINT, {sa_handler=0x5586bd16c380, sa_mask=~[RTMIN RT_1], sa_flags=SA_RESTORER, sa_restorer=0x7fe50f93b840}, NULL, 8) = 0
[... cut ...]
clone(child_stack=NULL, flags=CLONE_CHILD_CLEARTID|CLONE_CHILD_SETTID|SIGCHLD, child_tidptr=0x7fe50fac6850) = 15399
wait4(-1, [{WIFSIGNALED(s) && WTERMSIG(s) == SIGSEGV}], 0, NULL) = 15399
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_KILLED, si_pid=15399, si_uid=1000, si_status=SIGSEGV, si_utime=0, si_stime=0} ---
rt_sigreturn({mask=[]})                 = 15399
write(2, "Segmentation fault\n", 19Segmentation fault
)    = 19
read(10, "", 8192)                      = 0
exit_group(139)                         = ?
+++ exited with 139 +++

flindt, 2021-02-06:
LD_DEBUG=libs ./FreeJoyQt-x86_64.AppImage 
     16371:     find library=libdl.so.2 [0]; searching
     16371:      search cache=/etc/ld.so.cache
     16371:       trying file=/lib/x86_64-linux-gnu/libdl.so.2
     16371:
     16371:     find library=libpthread.so.0 [0]; searching
     16371:      search cache=/etc/ld.so.cache
     16371:       trying file=/lib/x86_64-linux-gnu/libpthread.so.0
     16371:
     16371:     find library=libz.so.1 [0]; searching
     16371:      search cache=/etc/ld.so.cache
     16371:       trying file=/lib/x86_64-linux-gnu/libz.so.1
     16371:
     16371:     find library=libc.so.6 [0]; searching
     16371:      search cache=/etc/ld.so.cache
     16371:       trying file=/lib/x86_64-linux-gnu/libc.so.6
     16371:
     16371:
     16371:     calling init: /lib/x86_64-linux-gnu/libpthread.so.0
     16371:
     16371:
     16371:     calling init: /lib/x86_64-linux-gnu/libc.so.6
     16371:
     16371:
     16371:     calling init: /lib/x86_64-linux-gnu/libz.so.1
     16371:
     16371:
     16371:     calling init: /lib/x86_64-linux-gnu/libdl.so.2
     16371:
     16371:
     16371:     initialize program: ./FreeJoyQt-x86_64.AppImage
     16371:
     16371:
     16371:     transferring control: ./FreeJoyQt-x86_64.AppImage
     16371:
     16371:     find library=libfuse.so.2 [0]; searching
     16371:      search cache=/etc/ld.so.cache
     16371:       trying file=/lib/x86_64-linux-gnu/libfuse.so.2
     16371:
     16371:
     16371:     calling init: /lib/x86_64-linux-gnu/libfuse.so.2
     16371:
     16371:     find library=libc.so.6 [0]; searching
     16371:      search cache=/etc/ld.so.cache
     16371:       trying file=/lib/x86_64-linux-gnu/libc.so.6
     16371:
     16371:
     16371:     calling init: /lib/x86_64-linux-gnu/libc.so.6
     16371:
     16371:
     16371:     initialize program: /bin/sh
     16371:
     16371:
     16371:     transferring control: /bin/sh
     16371:
     16378:     find library=libc.so.6 [0]; searching
     16378:      search cache=/etc/ld.so.cache
     16378:       trying file=/lib/x86_64-linux-gnu/libc.so.6
     16378:
     16378:
     16378:     calling init: /lib/x86_64-linux-gnu/libc.so.6
     16378:
     16378:
     16378:     initialize program: readlink
     16378:
     16378:
     16378:     transferring control: readlink
     16378:
     16377:     find library=libc.so.6 [0]; searching
     16377:      search cache=/etc/ld.so.cache
[... 94 lines cut ...]
     16379:     find library=libpcre.so.3 [0]; searching
     16379:      search path=/tmp/.mount_FreeJoRoLMTD/usr/lib:/tmp/.mount_FreeJoRoLMTD/tls/haswell/x86_64:/tmp/.mount_FreeJoRoLMTD/tls/haswell:/tmp/.mount_FreeJoRoLMTD/tls/x86_64:/tmp/.mount_FreeJoRoLMTD/tls:/tmp/.mount_FreeJoRoLMTD/haswell/x86_64:/tmp/.mount_FreeJoRoLMTD/haswell:/tmp/.mount_FreeJoRoLMTD/x86_64:/tmp/.mount_FreeJoRoLMTD:tls/haswell/x86_64:tls/haswell:tls/x86_64:tls:haswell/x86_64:haswell:x86_64:            (LD_LIBRARY_PATH)
     16379:       trying file=/tmp/.mount_FreeJoRoLMTD/usr/lib/libpcre.so.3
     16379:
     16379:     find library=libX11.so.6 [0]; searching
     16379:      search path=/tmp/.mount_FreeJoRoLMTD/usr/lib:/tmp/.mount_FreeJoRoLMTD/tls/haswell/x86_64:/tmp/.mount_FreeJoRoLMTD/tls/haswell:/tmp/.mount_FreeJoRoLMTD/tls/x86_64:/tmp/.mount_FreeJoRoLMTD/tls:/tmp/.mount_FreeJoRoLMTD/haswell/x86_64:/tmp/.mount_FreeJoRoLMTD/haswell:/tmp/.mount_FreeJoRoLMTD/x86_64:/tmp/.mount_FreeJoRoLMTD:tls/haswell/x86_64:tls/haswell:tls/x86_64:tls:haswell/x86_64:haswell:x86_64:            (LD_LIBRARY_PATH)
     16379:       trying file=/tmp/.mount_FreeJoRoLMTD/usr/lib/libX11.so.6
     16379:
     16379:     find library=libxcb.so.1 [0]; searching
     16379:      search path=/tmp/.mount_FreeJoRoLMTD/usr/lib:/tmp/.mount_FreeJoRoLMTD/tls/haswell/x86_64:/tmp/.mount_FreeJoRoLMTD/tls/haswell:/tmp/.mount_FreeJoRoLMTD/tls/x86_64:/tmp/.mount_FreeJoRoLMTD/tls:/tmp/.mount_FreeJoRoLMTD/haswell/x86_64:/tmp/.mount_FreeJoRoLMTD/haswell:/tmp/.mount_FreeJoRoLMTD/x86_64:/tmp/.mount_FreeJoRoLMTD:tls/haswell/x86_64:tls/haswell:tls/x86_64:tls:haswell/x86_64:haswell:x86_64:            (LD_LIBRARY_PATH)
     16379:       trying file=/tmp/.mount_FreeJoRoLMTD/usr/lib/libxcb.so.1
     16379:
     16379:     find library=libXau.so.6 [0]; searching
     16379:      search path=/tmp/.mount_FreeJoRoLMTD/usr/lib:/tmp/.mount_FreeJoRoLMTD/tls/haswell/x86_64:/tmp/.mount_FreeJoRoLMTD/tls/haswell:/tmp/.mount_FreeJoRoLMTD/tls/x86_64:/tmp/.mount_FreeJoRoLMTD/tls:/tmp/.mount_FreeJoRoLMTD/haswell/x86_64:/tmp/.mount_FreeJoRoLMTD/haswell:/tmp/.mount_FreeJoRoLMTD/x86_64:/tmp/.mount_FreeJoRoLMTD:tls/haswell/x86_64:tls/haswell:tls/x86_64:tls:haswell/x86_64:haswell:x86_64:            (LD_LIBRARY_PATH)
     16379:       trying file=/tmp/.mount_FreeJoRoLMTD/usr/lib/libXau.so.6
     16379:
     16379:     find library=libXdmcp.so.6 [0]; searching
     16379:      search path=/tmp/.mount_FreeJoRoLMTD/usr/lib:/tmp/.mount_FreeJoRoLMTD/tls/haswell/x86_64:/tmp/.mount_FreeJoRoLMTD/tls/haswell:/tmp/.mount_FreeJoRoLMTD/tls/x86_64:/tmp/.mount_FreeJoRoLMTD/tls:/tmp/.mount_FreeJoRoLMTD/haswell/x86_64:/tmp/.mount_FreeJoRoLMTD/haswell:/tmp/.mount_FreeJoRoLMTD/x86_64:/tmp/.mount_FreeJoRoLMTD:tls/haswell/x86_64:tls/haswell:tls/x86_64:tls:haswell/x86_64:haswell:x86_64:            (LD_LIBRARY_PATH)
     16379:       trying file=/tmp/.mount_FreeJoRoLMTD/usr/lib/libXdmcp.so.6
     16379:
     16379:     find library=libbsd.so.0 [0]; searching
[... cut ...] LMTD/usr/lib:/tmp/.mount_FreeJoRoLMTD/tls/haswell/x86_64:/tmp/.mount_FreeJoRoLMTD/tls/haswell:/tmp/.mount_FreeJoRoLMTD/tls/x86_64:/tmp/.mount_FreeJoRoLMTD/tls:/tmp/.mount_FreeJoRoLMTD/haswell/x86_64:/tmp/.mount_FreeJoRoLMTD/haswell:/tmp/.mount_FreeJoRoLMTD/x86_64:/tmp/.mount_FreeJoRoLMTD:tls/haswell/x86_64:tls/haswell:tls/x86_64:tls:haswell/x86_64:haswell:x86_64:            (LD_LIBRARY_PATH)
     16379:       trying file=/tmp/.mount_FreeJoRoLMTD/usr/lib/libbsd.so.0
     16379:
Segmentation fault

flindt, 2021-02-06:
Do you have a list of dependencies?

flindt, 2021-02-06:
How do I build the configurator? 
Qt creator does not recognize the .pro files.

Reksotiv, 2021-02-06:
I am very bad at Linux. This works on ubuntu and mint. I'll try to install debian 10 and figure out the problem

Reksotiv, 2021-02-28:
I updated AppImage in v1.7.0b2

closed 2021-05-17 by vostrenkov