FreeJoy-Team/FreeJoy #98: More question then problem

pipes80-collab, 2021-02-13:
Good morning, more than a problem is a question, would it be possible to use a rotary encoder instead of the normal potentiometers?  and if so, what difference could they have compared to potentiometers?

wizhippo, 2021-02-13:
They are supported, main issue is they do not know their initial position.

Another good alternative would be linear hall effect sensor as they do not wear like a potentiometer

pipes80-collab, 2021-02-14:
> 
    to put a hall sensor I would have to build a system to replace the potentiometer, as in some guides I have seen, otherwise buy hall sensors as per link but very expensive

https://www.racedepartment.com/threads/g27-g29-hall-sensor-mod.190696/

https://tomyracing.com/index.php?language=en&module=productsd&content=pedhall

wizhippo, 2021-02-14:
Yes you could use Sensors like https://www.amazon.ca/AH3503-UGN3503UA-UGN3503-Casing-Linear/dp/B07KPLVPCR and magnets to make your own or buy the more expensive hall effect pots that are drop in replacements