FreeJoy-Team/FreeJoy #100: Cannot connect on linux

e-wartal, 2021-02-26:
[30058.419960] usb usb1-port2: attempt power cycle
[30059.071876] usb 1-2: new full-speed USB device number 35 using xhci_hcd
[30059.072142] usb 1-2: Device not responding to setup address.
[30059.280137] usb 1-2: Device not responding to setup address.
[30059.487799] usb 1-2: device not accepting address 35, error -71
[30059.615752] usb 1-2: new full-speed USB device number 36 using xhci_hcd
[30059.615851] usb 1-2: Device not responding to setup address.
[30059.823981] usb 1-2: Device not responding to setup address.
[30060.031809] usb 1-2: device not accepting address 36, error -71
[30060.031937] usb usb1-port2: unable to enumerate USB device

Reksotiv, 2021-02-27:
Did you copy 99-hid-FreeJoy.rules file to /etc/udev/rules.d from linux release archive?

e-wartal, 2021-02-27:
Thanks for quick response, I have just copy that file, but still doesn't work, and I have the same issue, I have tried with different USB cable length and different socket and i have the same problem all the time, quite strange. There is also problem nn windows "Device Descriptor request Failed", maybe there is problem with cheap "hardware" ...

Reksotiv, 2021-02-27:
If device does not work in Windows, then the problem is hardware. Have you tried reflashing firmware?

vostrenkov, 2021-02-27:
https://github.com/FreeJoy-Team/FreeJoyWiki/issues/19

e-wartal, 2021-02-27:
I have 3 boards and none of them seems to be work, I have reflashed firmware but still have the same problem.

e-wartal, 2021-02-27:
Hardware: HW-267

vostrenkov, 2021-02-27:
> Hardware: HW-267

So what about R10 value?
http://amitesh-singh.github.io/stm32/2017/05/27/Overcoming-wrong-pullup-in-blue-pill.html

e-wartal, 2021-02-27:
Already saw this webpage, and changed R10 to 1.5K and still have the same problem.

vostrenkov, 2021-02-27:
Here is prebuilt ST's example project from official USB library. Are you able to run it?
[JoyStickMouse.zip](https://github.com/FreeJoy-Team/FreeJoy/files/6055201/JoyStickMouse.zip)

e-wartal, 2021-02-27:
Yeah, still the same issue, I tested in this same way before but without result, this is huge issue I see there are a lots of articles and youtube video about it, It looks like hardware issue. This board has also poor quality of soldering

pipes80-collab, 2021-02-27:
My arceli has a 251 resistor and, I don't know the value

reinholdPL, 2021-03-08:
had the same problem. then erased and flashed blue pill under windows 10 according to https://github.com/FreeJoy-Team/FreeJoyWiki/blob/master/eng/Flashing-firmware-with-ST-Link-programmer.md
blue pill started showing properly under linux as freejoy.