FreeJoy-Team/FreeJoy #101: filter level

pipes80-collab, 2021-02-28:
When I set filter level on 7 there is an huge input lag, it's normal? i can see on freejoy configurator

vostrenkov, 2021-02-28:
Of course, higher filter level is smoother but higher input lag

pipes80-collab, 2021-02-28:
Ok thanks for quick answer...can give problem put 1 ms on usb exchange period?

vostrenkov, 2021-02-28:
USB exchange period dont affect on filter level or input lag

pipes80-collab, 2021-02-28:
But it can improved?

vostrenkov, 2021-02-28:
What exactly you suggest to improve?

pipes80-collab, 2021-02-28:
I think of a speech of input lag, a slight improvement as happens with mice that are overclocked via usb at 1000hz and therefore with a bt interval of 1 ms

pipes80-collab, 2021-02-28:
> What exactly you suggest to improve?

I wanted to write that the change of pooling from 5 to 1 ms can improve the input lag, at least this happens with devices like mice

vostrenkov, 2021-02-28:
In freejoy usb exchange rate it not equal to ADC and sensors reading rate. Sensors are read once in 2 seconds so it is always 500Hz. This is maximum frequency that we can achieve with current software.
Why do you need so high filter level? Usually there are 3-4 levels are maximum that needed in case of good (or at least not bad) hardware. If filter level 7 is not enough for you it makes sense to improve your hardware design

pipes80-collab, 2021-02-28:
the pooling rate is the data exchange between the device and the USB port, in fact with the software for the oc of the USB ports and the low hid filter it is possible to bring the device to 1000 hz even if I read and acrivo everything says that the  btinterval is greater than 1 ms ... now I'm from the phone but as soon as I'm at the PC I send an image of R&W everything

pipes80-collab, 2021-03-01:
I want try to change potentiometer at my G29 pedals for see a things, Id fluctuation value can to be stopped or not I connect the new potentiometer directly at STM board

closed 2021-03-02 by pipes80-collab