FreeJoy-Team/FreeJoy #102: under linux every button works as button number 1

reinholdPL, 2021-03-08:
Firstly, I would like to thank you for all your work with Freejoy. A wonderful job.
I have problem on linux. When I run jstest-gtk (or jstest) I can see correct number of buttons and axes, but every button work as button number 1. It doesn't matter which button I press, always the button number 1 shows as pressed. FreeJoyConfiguratorQt correctly detects each button press.

Problem persists on both Ubuntu 20.04 and Debian 10.8. I am using the latest version of Freejoy v1.7.0.
I'm sure that everything is fine with electronics and configuration because device works 
properly on Windows 10 and Windows XP (each button works as it should).

Maybe it's a problem with usb hid implementation on linux? I will be grateful for any help.

Reksotiv, 2021-03-09:
Thanks for report. I confirm there is a bug. Latest working version on debian v1.6.2b1. Working on fix

Reksotiv, 2021-03-09:
yes

reinholdPL, 2021-03-17:
just wanted to confirm that this commit gets the job done and everything works correctly under linux now.

Reksotiv, 2021-03-17:
should works. This is an old known issue and we forgot to change it in 1.7. Changing 0x00 to 0x02 fixes the problem
[image]

Reksotiv, 2021-03-18:
fixed in [v1_7_0b2](https://github.com/FreeJoy-Team/FreeJoy/releases)

closed 2021-04-07 by vostrenkov