FreeJoy-Team/FreeJoy #104: About HEF4021BT  can not use CD4021 pin mapping

yzhanping, 2021-03-10:
Hi
Thank you very much to the Freejoy team. This is definitely a great software. I am trying to use it to DIY flight simulator. Its functions make me feel that DIY has great potential.
I encountered a small problem in use. The HEF4021BT used by Thrustmaster’s A10C controller has the same schematic pin definition as CD4021. I tried FREEJOY to drive this IC but failed. The buttons did not respond. I hope it works. Get help. Did I do something wrong?
The Freejoy STM32 motherboard works normally, and the interface uses 74HC165D+ software (shift register type: HC165 pull Up) to work normally

HEF4021B Schematic diagram
pin 3 - shiftreg_DATA
pin 10 - SPI_SCK
pin 9  - shiftreg_LATCH
pin 8 - GND
pin 16 - 3V3
[image]

CD4021 Schematic diagram
[image]

Freejoy SET
[image]
[image]
[image]

thanks for help~

yzhanping, 2021-03-10:
This is a DIY freejoy PCB, which links to the interface definition of Mapmaster; 
[image]

vostrenkov, 2021-03-17:
Hi! Just checked both 74hc165 and 4021 work with the latest release. 
Have you tried running pure shift registers config without any axes defined?

yzhanping, 2021-06-08:
I found the root of the problem. It was my fault. You must not use 5V as the power supply of CD4021. 3.3V can work normally, and 5V will burn. You can only replace the IC;

ernest-cpu, 2024-06-02:
cd4021b can support from 3v to 15 volt

> I found the root of the problem. It was my fault. You must not use 5V as the power supply of CD4021. 3.3V can work normally, and 5V will burn. You can only replace the IC;