FreeJoy-Team/FreeJoy #105: Use the AS5048A magnetic absolute position sensor as a multi-turn steering wheel

dr-milos, 2021-03-10:
This is a great way to build 900 degrees multi-turn steering wheel to drive Euro Truck Simulator 2.
Configuration file stays the same.
Support that AS5048A act as multi-turn steering wheel was added only on X-axe. (same sensor act as usual 360 degrees, or less, wheel on other axes)
Check the following video:
[https://youtu.be/zIPyCuZgbnk](url)
Only few lines of code was added into the "analog.c" source file in the section "else if (p_dev_config->pins[source] == AS5048A_CS)	// source AS5048A"
					```
                                     else
					{		
						raw_axis_data[i] = tmp16; //ocitana dvobajtna vrednost sa senzora se dodeli u niz sirovih vrednosti -ORIGINAL
						
						//Milos -izmenicu ovaj deo ako je u pitanju prva osa, da dodam podrsku za volan sa vise krugova za obrtanje, npr. 900 stepeni (delim sa 2)
						if (i==0) {
							raw_axis_data_Tmp[i] = tmp16; //za vise obrta AS5048A kod volana //originalne vrednosti su unipolarne
							if (raw_axis_data_Old[i] < 400 && raw_axis_data_Tmp[i] > 16000 ) iRotiranihKrugova[i]--;
							if (raw_axis_data_Old[i] > 16000 && raw_axis_data_Tmp[i] < 400 ) iRotiranihKrugova[i]++;
							raw_axis_data_Old[i] = raw_axis_data_Tmp[i];
							raw_axis_data[i] = (int16_t) ( raw_axis_data_Tmp[i] + iRotiranihKrugova[i] * 16383) / 2.0;  //bipolarno, tako pravim na izlazu
						}
```
Declaration of the new variables at the start of the file is:
```
int8_t iRotiranihKrugova[MAX_AXIS_NUM]; //za vise obrta AS5048A kod volana
uint16_t raw_axis_data_Tmp[MAX_AXIS_NUM]; //za vise obrta AS5048A kod volana
uint16_t raw_axis_data_Old[MAX_AXIS_NUM]; //za vise obrta AS5048A kod volana

```

ufocia, 2021-03-15:
Interesting. A multiturn trim pot would probably be easier to implement. You don't have to worry about the distance of the magnet  from the chip and no counting turns inside the code. Nonetheless, there are also benefits to the magnetic approach: easy reset, nearly infinite number of turns (limited only by the size of the variables), ... .

vostrenkov, 2021-05-17:
Multi-turn axes support is possible and not really hard to be implemented. We will take it into the account for the future work