FreeJoy-Team/FreeJoy #106: not problem but question

pipes80-collab, 2021-03-13:
is this https://github.com/ufocia/FreeJoy/releases problem solving for assetto corsa on linux?
i have try to flash but not detect freejoy

ufocia, 2021-03-13:
It fixes the issue of all buttons actions being conflated into button one actions. This is just the application without the boot loader. So the application needs to be flashed from the configurator. If freejoy has never been detected this will not help. If it was not detected the value of the resistors on the USB pins should be checked as some of the devices have the wrong resistor which cause them to not be detected. If the resistor is not a problem then the official hex file should be flashed with a hardware programmer and then the fixed version should be flashed using the configurator.

pipes80-collab, 2021-03-14:
that solved with flash from app qt thanks for help

closed 2021-03-14 by pipes80-collab