FreeJoy-Team/FreeJoy #107: Device not showing up in Windows after PC power on

p33t-r, 2021-04-03:
Hi, I sometimes have an issue - I leave the wheel plugged in the USB when PC is off, when I turn it back on, the wheel (Freejoy) doesn't show up in Windows or any game. Have to plug it out and back in again, then it shows and works properly. Happens on all of my USB ports. I even have a USB hub plugged in, Freejoy connected together with my Thrustmaster wheel and Arduino based HUD display with LEDs - all other devices show up and work every time after turning PC on, only Freejoy wheel has to be plugged out and in again almost every time, so I guess it's not my PC problem. Any idea what's causing that?

vostrenkov, 2021-04-03:
First of all specify which version of freejoy you use. This problem was actual for older versions

p33t-r, 2021-04-03:
1.6.2b3

vostrenkov, 2021-04-07:
please try the last 1.70 version