FreeJoy-Team/FreeJoy #109: Not working on X570 Chipset

spookeo, 2021-04-06:
Hi there,

I've been having an issue with setting up an STM32 with this, I was able to flash it with STM32 ST Link however, when using FreeJoyConfiguratorQT to set it up, I can't read or write - see https://github.com/FreeJoy-Team/FreeJoyConfiguratorQt/issues/10

However, I was able to configure it on a B550 Chipset computer, and I can see the device OK in game controllers, complete with custom name.

But back on my X570 machine (which is the machine I play on) it shows up only as FreeJoy v1.7.0, which leads me to believe my issue with the Configurator is deep down caused by this issue.

The weird thing is when I check with USBDView software, I can see the Device Name is indeed what I set as custom.

I've updated my motherboards BIOS and have also reinstalled the chipset driver to no avail.

Please let me know if you need any further information.

Thanks!

Reksotiv, 2021-04-07:
Hi
1)do you have issue with other USB devices?
2)have you tried [FreeJoy v1.6.2](https://github.com/FreeJoy-Team/FreeJoy/releases/tag/v1.6.2b4)? if not, could you check with two v1.6.2 configurators [old](https://github.com/FreeJoy-Team/FreeJoyConfigurator/releases/tag/v1.6.2b3) and [Qt](https://github.com/FreeJoy-Team/FreeJoyConfiguratorQt/releases/tag/1.1.2)? it works fine?

spookeo, 2021-04-07:
1- I don't believe so, I did install the latest BIOS to my motherboard which should fix any, but I am wondering whether I have something up with USB/HID drivers, perhaps that are causing some kind of conflict.. I'm curious to know if anyone else has used this with an X570 motherboard.

2- 1.6.2 seems to work ok with both configurators, however, the Device Name does not show correctly in my game controllers.

Reksotiv, 2021-04-07:
in 1.7 we changed USB communication, but I don't understand why it doesn’t work only for you.

> the Device Name does not show correctly in my game controllers.

https://github.com/FreeJoy-Team/FreeJoyConfigurator/issues/27 
1.7 configurator does this automatically when writing config

spookeo, 2021-04-07:
Oh right I see, well it seems to be working ok on 1.6.2 and I've sorted the naming issue. Would be great if I can get the latest version working but I have a feeling I'll need to reinstall Windows for that :D thanks a lot for your help so far.

Handsome1080P, 2024-11-04:
x570 lost USB ID after replug USB cable to a new port .And If it power on long time,board will stucked,need repower on .