FreeJoy-Team/FreeJoy #111: Linux Problem

gokhanhas, 2021-04-14:
First of all, I would like to thank those who contributed to developing such software and publishing it as open source.

My problem is as follows. When I connect the device in the QTCOnfigurator app, its name comes on, but disconnect and constantly in the console - Open HID device №1. VID "0483", PID "5757", Serial number "" - returns error. While there is no problem in WIndows, there is this problem in linux.

In addition, I can see the device in "jtest-gtk" application. The settings are working properly. However, ETS2 does not appear as a controller, for example. ETS2 also appears when I plug in my own gamepad. Again I can play on Windows without any problem.
I am using Ubuntu 20.04 LTS. UPD. v1_7_0b2.hex and UPD. I tried the v1_7_0b3.hex files separately. Is there anything else done for Linux other than uploading .hex files? What's the point I'm missing? Thanks.

Reksotiv, 2021-04-14:
Hi. https://github.com/FreeJoy-Team/FreeJoyConfiguratorQt 
[image]

gokhanhas, 2021-04-14:
Thanks for the very quick reply.  Unfortunately I haven't read it fully.  After doing, the QT app is now connected without any problems.  However, for example, ETS2 did not recognize the game as a gamepad.  Is there anything you can fix about this problem?  Thank you very, very much.

Reksotiv, 2021-04-14:
I cannot run games in a virtual machine. Don't know why it doesn't work. Maybe later i'll try to install it on main PC.

gokhanhas, 2021-04-14:
I will try to download other games right now.  Thanks again for your help.  I am using the system as dual boot.

Reksotiv, 2021-04-16:
I checked. ETS does not see the controller if output is turned off for X and Y axes

gokhanhas, 2021-04-18:
Hello, I just saw the message. Sorry. Does linux see in other ways? I use the X, Y and Z axes. I use three potentiometers.

Reksotiv, 2021-04-18:
Hi. You are right, only the axes are not enough for linux. You need to add at least 1 button
[image]

gokhanhas, 2021-04-18:
Thank you so much.  I will find a button and try it as soon as possible.

Reksotiv, 2021-04-18:
you don't need a physical button, just add Button GND/VCC in the configurator on any pin and set logical button

gokhanhas, 2021-04-18:
Thanks, very good then.  I will try immediately.

gokhanhas, 2021-04-18:
I add button by using configurator and then it will works. I played ETS2 in linux. Game see the STM32 by gamepad. Thanks for your helping.

Reksotiv, 2021-04-18:
you are welcome

closed 2021-04-18 by gokhanhas