FreeJoy-Team/FreeJoy #115: 74HC165 pull down give wired results with 5v vcc

Jack-XHP, 2021-05-05:
Although there is a line in the wiki that suggests 3.3v as Vcc, I was surprised to see 74hc165 with completely wrong behaviors when connected to 5v. For example, with just one 74hc165, no input is detected; with 4 of them chained together, every input in the first one will trigger all 32 inputs, the second one, each trigger every other input, the third one triggers two inputs together, and the last one cannot be detected.

If such wired behaviors are expected for 74hc165 with 5v Vcc, I think it is insufficient for just a line embedded deeply in the wiki. For use-cases with PCB, it will just render the whole board is useless just due to the Vcc issue.

vostrenkov, 2021-05-17:
SInce STM32 have 3.3V logic levels it should not work correctly with 5V logic periphery. 
Though STM32 have 5V tolerant pins it only means it can survive at **reading** 5V input signals but it still provides 3.3V output signals. In some cases 3.3V is not enough for HIGH logic level on the periphery (shift registers, sensors, etc) and it wont work correctly

virtual812, 2021-06-13:
@Jack-XHP You make some good points.
Whilst the issue is common knowlage to one group it's not so commonly known by others.
The STM32 whilst 5v tolerant in some cases on some pins is really best run at 3.3v.
I'd suggest running your entire setup at 3.3v

TomHVR, 2022-02-05:
hmmm I've having a similar problem with 4021 pull downs.

I originally set up the shift registers with freejoy and they were working as expected, then had to move my build over to mmjoy2 as my hall effect sensors couldn't run at 3.3v, Now with new AS5600 sensors I'm moving back to freejoy im encountering similar strangeness, running it over the 3.3v line. I even tried sending 5 volts and then running the dataline with a voltage divider to only send 3.3v into the pin on the bluepill but no change.

I can take the entire SR setup and plug it back into my mmjoy2 board and it works correctly.