FreeJoy-Team/FreeJoy #117: Question: Digital/Binary Multi-Wire signals

RossosHEX, 2021-05-08:
Hello! 
I am in the process of trying to hook up a real car Indicator Level to FreeJoy.
So far the reverse engineering produced me these results:
[image]

I have successfully bind the Headlights/Beam/Indicators/Wipers level functions as they appear to be operating as analog Axels.
Config file: 
[FreeJoy v1.7.0 mk4.cfg.txt](https://github.com/FreeJoy-Team/FreeJoy/files/6444860/FreeJoy.v1.7.0.mk4.cfg.txt)

At the moment I am stuck because I do not know how to connect the 2nd lever.
It has digital outputs on the pin connectors, High or Low.
Any suggestions as to how I can connect/configure the digital multi-wire interface?

Thank you in advance.
Any help would be highly appreciated.

vostrenkov, 2021-05-17:
Hi!
I'm not sure it is possible but you can try to do this with shift modificators chaining. Since shifts are logical buttons it is possible to set shifts on shifts (but it can work not very stable).
I pin image how it can theoretically work but I didnt try that on the hardware
[image]
[image]

RossosHEX, 2021-05-30:
Greetings!
Finally got time this weekend to tinker.

First, thank you for your reply. I did not end up with your proposed solution but it made me utilize the SHIFT modifiers. Thank you for that. I have set up the pin 10 and pin 11 as normal buttons, and then set up pin 9 as a SHIFT modifier. So it ended up being pin10 -> 1a, and pin10+pin9 -> 1b. Set up pin11 in similar fashion, pin11 -> 4a and pin11+pin9 -> 4b (as per the first Figure table).

Second, I have tried more variations of contacts, with which I was able to figure out positions 2 & 5 from first Figure I posted in the original post. Ended up connecting pins 6 and 8 and setting them up as Inverted Button. 

Here's the config file with which things are all working:
[FreeJoy v1.7.0 mk6.cfg.txt](https://github.com/FreeJoy-Team/FreeJoy/files/6566209/FreeJoy.v1.7.0.mk6.cfg.txt)

I would like to thank you for making this amazing project, and I would like share the photos of my project with you.

[image]

[image]

[image]

Feedback:
1. It's a shame the PA8 and PA9 when used as Fast Encoder utilize one of the Axes, rendering one of the Analog Inputs not utilized as analog axis.
2. For some reason when Disabling a Button in the Button config of the FreeJoy Configurator - the button right below it shows activation on the Disabled button and the rest of the buttons below are shift upwards as well. I've set the Shift button at Logical button 128 to avoid this. Is this intended behavior? (FreeJoy Configurator v1.7.0b3)
3. Wishful thinking: Looking forwards for the ForceFeedback at some point in the future.

Reksotiv, 2021-06-01:
> 2\. For some reason when Disabling a Button in the Button config of the FreeJoy Configurator - the button right below it shows activation on the Disabled button and the rest of the buttons below are shift upwards as well.

Fixed in https://github.com/FreeJoy-Team/FreeJoy/commit/5ce9d37b6e02c64e78b981e0a9d2a9fd76b2799e Thanks for feedback. Will be released in the next version