FreeJoy-Team/FreeJoy #120: The reason of encoder being interfered

uwlftoe17, 2021-05-17:
I have been using freejoy QT 1.70b2 hex. I found that when the switch box I made was used with fanatec dd1 or DD2, the encoder would be interfered, and I could see it dancing in the computer test window. I have four encoders. It is possible that fanatec DD has very strong electromagnetic interference.

However, when I use OSH software configuration code, the same thing doesn't happen. Of course, I use the same STM32 c8t6.

I'm guessing, is it because freejoy QT enables pull-up resistor??

I hope to get your help. Thank you

eviltheknivel, 2021-05-19:
Yes it gives problems with fanatec DD and some Wheels. Use a 3D printed Adapter to make sure that there is  no metal connection between Base and Wheel.
You can test it easy, remove the Wheel from the DD and connect it with usb. when buttons and all works fine then you know the problem.

jimbraj, 2021-12-29:
you can also ground the pcb to the wheel base and it will solve this issue.