FreeJoy-Team/FreeJoy #121: Matrix Button Rows Keep Failing

speedbird5, 2021-05-18:
Hi I have an STM 32 see attached pic.

I have flashed the device successfully with FreeJoy_1_7_0b3.hex and FreeJoy_1_7_0b3.bin using the ST-Link V4 software.

I am running FreeJoyQt.exe version 1.7.0b3

I have a matrix of 8 columns and 10 rows and I have a diode correctly placed between each switch.

The setup worked initially but after a while, two of the rows failed.

I swapped them to another free terminals  and re configured the new rows and a little while later the new rows failed.

The trouble is that I have run out of good rows.

It seems that it sees the two rows internally as the same row and causes all the buttons on those rows to illuminate.

It always seems to be the rows, not the columns.

[image]

speedbird5, 2021-06-17:
Hi,

further to my problem above, I think that it is caused by static electricity discharge on the case enclosing the STM32 which is damaging the board. 

Does anyone know how to protect the STM32 from electrostatic discharge?

Thanks

vostrenkov, 2021-06-27:
Hi, you can use ESD protection IC like this one:
[image]

speedbird5, 2021-06-27:
I have used 32 pins on my STM 32. How many of these ICs do I need to protect the board ? I am unsure how I would wire them up. 

Thanks

vostrenkov, 2021-06-27:
It can make sense to use TVS diodes to GND on each input for protection. They are quite cheap and can be easily wired

speedbird5, 2021-06-27:
Hi, so I need 32 of these ICs ?

vostrenkov, 2021-06-28:
I think it is better to use something like this on your pins:
https://www2.mouser.com/Circuit-Protection/ESD-Suppressors-TVS-Diodes/_/N-jy72?P=1yh8kcrZ1yh8nq3Z1yh8nq2Z1yh8nr5Z1y83ebpZ1yh8o3m

Unfortunately I'm changing some pinout for Controller Lite so it is still not available for purchasing