FreeJoy-Team/FreeJoy #123: [FR] Add support for classic controllers (use as a USB adapter)

eadmaster, 2021-05-29:
I'd like to suggest adding support for reading some protocols used by classic consoles controllers.

Some code can probably reused from existing projects like [this](https://github.com/mcgurk/Arduino-USB-HID-RetroJoystickAdapter/).

I am especially interested in systems with native 3.3V logic, that can be interfaced directly to an STM32 without the need of level shifters: PS1, N64, Gamecube and Wii.

Since most of these are serial protocols, multiple controllers could be connected at the same time.

vostrenkov, 2021-06-16:
I'd say this has quite low chance to be implemented because of slightly different project specialization. Thats fine that people are thinking of using freejoy in such applications but main goal is to make a platform interfacing with sensors/discrete inputs itself

eadmaster, 2021-06-16:
No problem, can i leave the issue opened for  future contributors  reference?

vostrenkov, 2021-06-16:
Sure