FreeJoy-Team/FreeJoy #126: [FR] Support for NXP GPIO Expander, eg: PCA9555

virtual812, 2021-06-13:
I've just become aware of a family of chips from NXP which seem like shift registers, but on a while new level.
They are bi-directional, and the family supports from 4 to 40 IO pins on a single IC.
May of these also include internal 100K pullup resistors.
The example i'm interested in is the PCA9555
About AUD$3 maybe closer to $2 in US currency.
Seems to be readily available.
Pretty attractive having something about the size of 1 shift register (maybe slightly larger) but with double the inputs and no need for pullup resistors.
I'd be curious to know what the FreeJoy devs think of this.

vostrenkov, 2021-06-16:
Yes, it is known good solution.
This may be implemented somewhen

dmberrycom, 2021-06-29:
I would suggest that whilst these  chips do offer a high density of ports, it would be better from an accessibility perspective to implement support for something like the  MCP23017 https://www.microchip.com/wwwproducts/en/mcp23017.

These chips are widely available and commonly used with lots of breakout boards easily available on amazon, AliExpress and common maker stores. 

https://www.waveshare.com/mcp23017-io-expansion-board.htm
https://wiki.dfrobot.com/Gravity:%20MCP23017%20IIC%20to%2016%20Digital%20IO%20Expansion%20Module%20SKU:%20DFR0626

virtual812, 2021-06-30:
> 
>something like the MCP23017 https://www.microchip.com/wwwproducts/en/mcp23017.
> 

That looks even better, more accessible, variety of great packages and internal 100k pullups.
I really do agree, this would be a great option.
Shift registers are OK, but once you have a few of them and have to add the associated pull up resistors things get messy fast.

virtual812, 2022-07-07:
Hey guys, wondering if there is anything i can do to help with this.
No idea how to write code, but can we send someone on the team some dev boards with these these chips?
Beer money?

Interested in either or both the PCA9555 and family or MCP23017