FreeJoy-Team/FreeJoy #128: POV Center is also triggerd after POV Up is released

ethan54013, 2021-07-11:
Hi, hope you are doing well! 
I'm using ALPS RKJXT1F42001 as the POV input, and I've seen people reporting this issue (POV Center is also triggerd after POV Up is released) in other threads, but after trying setting the debounce delay to 100-150 as you suggested, the problem is still there. Only if you quickly press and release the Up direction, the problem will disappear, but it is very difficult to control. So I want to ask if there is any alternative solution to this? Thank you!

evolaco, 2021-12-11:
hello, have you found a solution?

vostrenkov, 2021-12-12:
> hello, have you found a solution?
Yes, it is pushed in /dev and will be released soon