FreeJoy-Team/FreeJoy #132: Same pins not working right CH32 clone blue pill

cagri-yuksel, 2021-08-19:
Hello i got CH32 label clone blue pill. Everyting working well currently using FreeJoy_1_7_0b3 version. 

Here problem B4 B1 and B0 pins when i try to use as button just blinking randomly. 

Maybe this is problem with card maybe software not compatible with clone bluepill, i check all short circuit everyting looks fine.
Do you think look clone cards and support them? Finding real st chip in market is really hard.

Thank you for hard working.
[image]

vostrenkov, 2021-08-22:
Hey
Unfortunately there it a big variety of STM chip clones. Some of them works just fine but the others have random issues. It is quite hard to add support for that

cagri-yuksel, 2021-08-22:
Thank you for answer sir,  you are already told it's a hobby project. Thank you for your hard working here.

closed 2021-08-22 by cagri-yuksel

vostrenkov, 2021-08-22:
After fixing major bugs of v1.7.0 I gonna port this project to STM32F401 chip. There are not that many clones for it and chips are easier to get

cagri-yuksel, 2021-08-22:
It's a amazing news. Also cheaper than bluepill.