FreeJoy-Team/FreeJoy #135: When disabling special funtion "equal"  on encoding axis using a toggle function, please retain last equal value instead of resetting to last known value

recoilfx, 2021-08-23:
I built a single lever throttle, and have an encoder as a second throttle axis. 

The encoder axis has the special function "equal", toggled by a switch button. So by default, the encoder axis always have the same value as the main lever axis, and when the toggle switch is off, I effectively have 2 throttle axes. 

However, currently, if I toggle the switch off, the value of the encoder axis reverts to last known value before having the equal function on, which can be a huge change when compared to the last equal value (and sending my plane into a tail spin while I frantically rotate my encoder axis to the correct throttle value). Can we have an option where it simply retains the last equal value when disabling the special function?

Thank you so much!

vostrenkov, 2021-08-23:
> hmm, I think I can add this

no so easy though..

recoilfx, 2021-08-24:
I see. Well, it would be awesome if it's implemented. This allows users with single lever throttle to emulate dual throttle axes with an encoder in the rare occasions when one engine goes wrong. Thank you for all your hard work!