FreeJoy-Team/FreeJoy #151: bug report about shift registers V1.73

piogrammer, 2021-12-31:
i have tryed to add 4 hc165 ,pin map  shown in photo below
[image]
but they cant work 
and i found sothing wrong 
[image]

as you can see 
all the latch pin are the same one（whitch i selected first）
i selected b12 b13 b14 b15 as Latch，then all the latch pin are B12
i selected b15 b14 b13 b12 as Latch，so  all the latch pin are B15
 sincerely hope that you can repair this fault as soon as possible.
love you dude

Reksotiv, 2021-12-31:
I fixed this many months ago for version 1.7.1, but we delayed the release. This is an adapted version for 1.7.0, try it
[1.7.0.4-test.zip](https://github.com/FreeJoy-Team/FreeJoy/files/7795206/1.7.0.4-test.zip)

piogrammer, 2022-01-06:
I think you can release the source code now,
It's exciting that this version of the register works normally.
thanks dude

closed 2022-01-11 by vostrenkov