FreeJoy-Team/FreeJoy #152: [Question] TLE5011 and further questions

Amoksepp, 2022-01-04:
Hello FreeJoy Community,

first i have to say thank you for your great project.

A little warning, i have not much experience with this topic, i did some testing with hall effect sensors and a arduino yet.
I read about FreeJoy and TLE5011 sensors and bought them, but i m overhelmed by the complexity of the topic.

So i have a few basic question for better understanding.

  1.  Where ist the best place to ask questions? Do you know a discord or something similiar with active members?
  2. the TLE5011 is supported by the freejoy board, so i just need to do the wiring and configuration with the configurator to get it running (beside fine tuning positioning....), is this correct?
  3.  about the wiring, i need the PCB form https://github.com/FreeJoy-Team/FreeJoyWiki/tree/master/3rd-party/hardware, let it produce by a PCB factory, solder the SMD components and the TLE5011 to it and wire the pins to the according Freejoy pins?

Thank you in advance.

aguLepe, 2022-01-06:
1.- There is a discord: https://discord.com/invite/szqaJE7 Channel freejoy.
2.- That's correct. I think you need to design a PCB or use the edsing in /3rd-party/hardaware or use a tiny sop8 to dip8 breakout board.

vostrenkov, 2022-01-11:
1. Right, you gonna get answers faster if ask in discord. Freejoy channel in HOTAS discord is official
2-3. TLE5011 is fully supported. You should wire it as described in [wiki](https://github.com/FreeJoy-Team/FreeJoyWiki/blob/master/eng/TLE501x-connection.md) in any way you like (proto-pcb, preassembled module or custom pcb)