FreeJoy-Team/FreeJoy #153: Question regarding axis switching

wasyl00, 2022-01-10:
I'm in the process of building farming sim controller. I would like to know if its possible to have one 2axis joystick but use it to control 4 axis with use of switch button between them. Basically switch in position A - joystick controls axis 1 and 2. Switch in position B joystick controls axis 3 and 4.  I'm considering hardware switching and software. I see multiple issues when trying to do it in hardware and I'm looking around for software options. Thanks

vostrenkov, 2022-01-11:
What do you want to have on output of inactive axis?

wasyl00, 2022-01-11:
Default positions for 2 innactive axis or even last values before the switch as I can always release joystick to conter itself before switching to other axis.

vostrenkov, 2022-01-11:
You can use "Equal" function for axis activated by the button.
If you add auxiliary axis with fixed value on input (tied to GND, VCC or using voltage divider) you can force your main axis equal their values to this secondary axis by button press.
You can also use only one physical button implementing two logical button inverted related each other to control described functions of the axis
[image]

closed 2022-02-21 by wasyl00