FreeJoy-Team/FreeJoy #159: Questions about adding mpu6050 and ads1118

yzhanping, 2022-03-01:
Hello,

I have two questions to ask:

1. Have you considered adding mpu6050 components? It has three axes and acceleration. There is a general PCBA on the market. The connection mode is I2C with 16bit accuracy. It seems to be a cool component. Use it for X, y and Z axes;It can be used for head chasing, and the attitude sensing is very good;
[image]

2. Have you considered using ads1118 as the original of digital to analog conversion? It is SPI connection and its performance is superior to I2C
[image]

Thank you.

vostrenkov, 2022-03-01:
1) Accelerometers and other mems are not planned to be added to freejoy. Any reason to do that?
2) I'll take a look at this

FlyingPotato-131, 2022-05-05:
An accelerometer/gyroscope would be extremely useful in creating, say, 6 axis controllers