FreeJoy-Team/FreeJoy #160: Noisy rotary encoders

elbosmo, 2022-03-05:
I've made a simple button box with 3 axis and works very nicely and was easy to do with FreeJoy and the nice config tool.

The only issue I have is with a rotary encoder. 
I have tried two different models and multiple of the same model. 
I have tried the 1.7.1b0 and 1.7.0b3 firmware as well.
Also I have tried to use different ports on the STM as inputs (C13/14 and B11/B10), in all cases directly connected with "Button Gnd" configuration.
I have tried the 1x, 2x and 4x encoder settings.
Also tried stronger pull-ups and capacitors etc, but made no difference.

In all cases the encoder does not work without occasional false presses. When rotating one direction, it randomly also has a button press for the other direction, same for rotating the other way. This happens say each full rotation at least once. Rotating slow or fast does not make a difference.
It may also have multiple presses while physically only moving it a single notch.
It may generate a button press but rarely, when just wiggling knob without passing a notch

I have tested the same encoder with https://github.com/mathertel/RotaryEncoder with an arduino nano.
I tested the simple SimplePollRotator example using the "FOUR3" latchmode and it worked without any problems, no false or double readings at all.

beastieboy, 2022-04-20:
I tested FreeJoy encoder code in separate project. It works but pin configuration had to be changed to PullDown. Most of encoders use different polarity than assumed here.

leandrol360, 2022-10-05:
@beastieboy, how can i change the pin to pulldown?

Merlyn42, 2022-10-19:
I am having a similar issue although I haven't tested as thoroughly as @leandrol360.

leandrol360, 2022-12-06:
I'm still having the problem, does anyone have any tips?

Stout001, 2025-02-09:
I have no solution for you but I also have sever problems with encoders in FreeJoy. It's a pity that they are not working and randomly "roll back". That makes it useless for a wheel.

RMSSimRacing, 2026-05-08:
> I don't really know how to describe it any more than I did. If I turn the
> encoder one way it works, but if I then turn it the other the first turn
> will be ignored.
>
> CW CW CW
> works fine
>
> CCW CCW CCW
> Works fine
>
> CW CCW(ignored) CCW
> The first counterclockwise turn is not detected.
>
> These are 2x encoders

Okay, thank you.
What debounce and encoder press time are you using?
Are you using the EC11 encoder?
How many pulses does it have?
After your message, I repeated several tests with my EC16 encoder and
observed no losses.