FreeJoy-Team/FreeJoy #162: Fake blue pills

andyweed, 2022-03-09:
Hi guys,

First of all, huge thanks for your work, FreeJoy is awesome! :)
Just a quick question rather than an issue: what's your opinion on fake blue pills? I mean some of them I can load your firmware onto, some of them I can't, and honestly I can't tell the difference between fake and genuine chips.
Do you guys know a method by any chance to tell the fake ones apart from the genuine ones?

Many thanks.

Reksotiv, 2022-03-09:
Hi
I have 3 bluepilles, two of them are fake but work like original. [About fakes](https://github.com/keirf/Greaseweazle/wiki/STM32-Fakes)
And you can check with this application:
[stmTest.zip](https://github.com/FreeJoy-Team/FreeJoy/files/8214079/stmTest.zip)

[image]

andyweed, 2022-03-09:
That's great, thank you very much.
I have a few which I can load the fw onto, but the FreeJoyQT does not recognize them, and I have no idea what I'm doing wrong. They are all genuine according to the tester.

andyweed, 2022-03-09:
Got it fixed, thank you, I got confused with VID numbers :)

Many thanks for your quick help!

closed 2022-03-09 by andyweed

harrinator-yxz, 2025-02-18:
Will anything that says STM32F103C8 work?