FreeJoy-Team/FreeJoy #163: Dual paddle clutch with bite point

luowilliam326, 2022-03-23:
I tried a few ways of combining axis to try to get this function but failed. Just want to see if it's possible at all.
Basically I want to have 100% input on pot 1 and use pot2 to add to or subtract from pot 1, that output would be the bite point. Then at race start, I would have 100% input on pot 1 and pot 3, at lights out, I drop input from pot3 to 0% leaving the total output at pot1+Pot2, I then slowing release pot1 back to 0%.
I think I can get 2 pots to combine no problem, but to have a thrid pot modifying the first pot and have first pot be able to control the combined output of pot 1 and 2 is difficult.

vostrenkov, 2022-03-26:
@Reksotiv have you ever managed to set up such configuration?

Reksotiv, 2022-03-26:
No, i didn't make a clutch

luowilliam326, 2022-04-01:
@vostrenkov I think it's a very common feature for people making their own wheel. Do you think you can add it?