FreeJoy-Team/FreeJoy #169: 12 Positionen Rotary Switch

dillario, 2022-05-02:
Hello. 

I would like to connect this Rotary Switch. Only I can't find anything about it in the instructions. Can you help me with that? 

Thank you 

Greets 

https://www.mouser.de/ProductDetail/NKK-Switches/MRK112-A?qs=vtN2KNUHTpVD9s4cK7LgDw%3D%3D&mgh=1&vip=1&gclid=CjwKCAjwgr6TBhAGEiwA3aVuIYnO_6uvMpYZq5bPs0kFcWu8GzfI0XJ4OunCKIbTrRA6wzF8d_3FsRoCMREQAvD_BwE

Reksotiv, 2022-05-03:
you can connect like buttons(single, matrix)
[image]
or make a potentiometer by soldering resistors and connect as [Axis to Buttons](https://github.com/FreeJoy-Team/FreeJoyWiki/blob/master/eng/Axis-to-buttons-function.md)
[image]

dillario, 2022-05-03:
Thank you. If I do it like in the image below, is it like an encoder? And how do I do it if I want to connect 3 of them?

dillario, 2022-05-03:
Super. I will test it. Thank you very much

dillario, 2025-11-07:
> [@Reksotiv](https://github.com/Reksotiv) will it work with one switch or two must be used?

Yes. Thanks 😊