FreeJoy-Team/FreeJoy #172: Axes not reporting positions properly

gljones2001, 2022-05-27:
Hello there, this is a weird one.

When I flash the FreeJoy firmware to my STM32 device, the axes (Yes, axes IS the plural of axis) all show up in https://gamepad-tester.com/ with values of `0.99998`.

[image]

As you can see, this happens with all axes. This does not change, no matter how I configure the axes. When I connect my hall sensor joystick to the board and move it, the value changes to one of the following:
 - `1.00000`
 - `0.99999`
 - `0.99998`
 - `-0.99998`
 - `-0.99999`
 - `-1.00000`

Those are the only values I can get it to show. I think there is something wrong with either my board, or the firmware. IDK.
Any ideas?

Thanks, 
G

vostrenkov, 2022-05-28:
Please pin here your configuration file and your connection diagram. It is hard to make any guesses basing on current information

Jesse-Pinkman-443, 2022-05-29:
Hi Yuri, I will once I have further designed my devices.

Best regards
Jesse

exilaus, 2022-06-20:
@gljones2001 try to configure Xoutput  https://github.com/csutorasa/XOutput for wrapping directinput API like freejoy to last API called Xinput. i resolve my issue with new games but create some limitation because Xinput from MS reduce number of button and numers of axis (but standardize output for him console and pc.