FreeJoy-Team/FreeJoy #193: Firmware v1.7.1b3 (latest) freezing on STM32F103C8 Revision Z

prikrylm, 2023-01-04:
FreeJoy firmware freezing after aprox. 10s of workig on an STM32 with 256 kB Flash memory.
Programmer reports this CPU as Revision Z, ID 0x414. Maybe is there some issue with emulation of EEPROM?

YifeiZhao0816, 2023-02-26:
Same issue here. Also tried v1.6.0, the board also freezes

Stout001, 2023-03-22:
> > Mismo problema aquí. También probé v1.6.0, la placa también se congela
> 
> Hi, could you fix your problem? In my case it does not recognize hall sensors or potentiometers when configuring axes, I do not know why, I test the same sensors in arduino Leonardo and they work well. I use v1.7.1b3 I consult because I am new to this, in case I want to install a previous version how should I proceed?? Install the firmware on STM32 using ST Link v2.

Could be that you are using a fake STM32F103C8 and not a genuine. I have several fakes most of them (not all) of them behaves differently. Either not connecting, conencting and disconnecting intermitently or sems to work with buttons but not with hall sensors or potentiometers. I read somehere that some fakes runs OK on USB2 but not USB3 and most are 64K
Very frustating.