FreeJoy-Team/FreeJoy #198: Help Sensor hall !!

Sprintel82, 2023-03-01:
Hello, I need help, I have simulator pedals, previously I had them connected with arduino and a sketch of my own. Now I want to change the electronics to stm32 and freejoy. But it happens to me that connect the hall sensors as indicated by the freejoywiki, they do not work, they do not give any type of signal, it is strange because with arduino they work.

 Greetings, thanks and sorry for my English.

nopain61, 2023-03-02:
you can use it   if it have a analog pin so hallsensor you can connect it because ı made it a shifter  with   mlx90333  ı used analog pin and 3.3 and gnd it is very basic after  ı calibrated to  on freejoy configurtor but ı  get error very interesting error 
also you can use bad board ı  tried a card    freejoy is not work  maybe this link you can help you  
https://github.com/FreeJoy-Team/FreeJoyConfiguratorQt/issues/30  you buy 64 kb ,ıt has a full close metal button

Sprintel82, 2023-03-03:
> you can use it if it have a analog pin so hallsensor you can connect it because ı made it a shifter with mlx90333 ı used analog pin and 3.3 and gnd it is very basic after ı calibrated to on freejoy configurtor also you can use bad board ı tried a card freejoy is not work maybe this link you can help you [FreeJoy-Team/FreeJoyConfiguratorQt#30](https://github.com/FreeJoy-Team/FreeJoyConfiguratorQt/issues/30) you buy 64 kb ,ıt has a full close metal button but it is not one circle button should be square button

I could make the hall sensors work, but now I can not get the load cell for the brake to work. I will attach the diagram.
Do not connect it exactly the same, do not use rj45, I use 4-pin gx16 connectors, and connect red cable to v+, white to -vin, green to
+vin and black to gnd.
What it does when I connect is oscillate in the middle, very little, but when I press the pedal it does nothing. I'll try resoldering the connectors, maybe that's the problem, I don't use rj45 because I don't have clips to crimp.

nopain61, 2023-03-03:
bro you should buy one  ina122p amplifikator but    all of ina122p  is buying in the aliexpress as  broken :) so  ı bought ina128p  their are working  ı am using ı bought this link    its are woriking  https://www.aliexpress.com/item/4000098403942.html?srcSns=sns_WhatsApp&spreadType=socialShare&bizType=ProductDetail&social_params=20850354961&aff_fcid=ff3e02e153494b09b0c81c9dddbf451f-1677830065179-07753-_mMngyFY&tt=MG&aff_fsk=_mMngyFY&aff_platform=default&sk=_mMngyFY&aff_trace_key=ff3e02e153494b09b0c81c9dddbf451f-1677830065179-07753-_mMngyFY&shareId=20850354961&businessType=ProductDetail&platform=AE&terminal_id=7fe2ee72690243ceb3e491c1920e8ced&afSmartRedirect=y

Sprintel82, 2023-03-03:
> bro ,for loadcells; ı use full loadcell gas brake clutch and handbrake so you should buy one ina122p amplifikator but all of ina122p is buying in the aliexpress as broken :) so ı bought ina128p their are working ı am using ı bought this link its are woriking https://www.aliexpress.com/item/4000098403942.html?srcSns=sns_WhatsApp&spreadType=socialShare&bizType=ProductDetail&social_params=20850354961&aff_fcid=ff3e02e153494b09b0c81c9dddbf451f-1677830065179-07753-_mMngyFY&tt=MG&aff_fsk=_mMngyFY&aff_platform=default&sk=_mMngyFY&aff_trace_key=ff3e02e153494b09b0c81c9dddbf451f-1677830065179-07753-_mMngyFY&shareId=20850354961&businessType=ProductDetail&platform=AE&terminal_id=7fe2ee72690243ceb3e491c1920e8ced&afSmartRedirect=y

I use ina122, trimpot sermet 1k, 0,2uf capacitor. How to know if ina is not working well ??

nopain61, 2023-03-04:
if it is not working which about  axis  absolutely  ina is  broken  :)   so  you are going to buy ina128  ı  give you  seller link