FreeJoy-Team/FreeJoy #200: Can we expect an update to F401/F411 Blackpill?

Stout001, 2023-03-22:
I have so many Blupills that are fake and not working as intended. I read somwhere (IL-2 RU?) that there are plans for a change over to STMF401/411. Is this something that will be announced in the near future? 

Just wondering because I'm on my way to make double STM32F103ct86 shield but will wait if a new version of Freejoy is imminent.

lijokdb, 2023-04-29:
bro stm f401  not work struck sume %  iam use emc in this bord but not  feejoy  ithink bugs 
[image]

Stout001, 2023-04-29:
F401/411 is not compatible with Freejoy. You have to use STM32F103C8T6 "bluepill" until it's time for a new board. That was what I was wondering when it'll be.