FreeJoy-Team/FreeJoy #203: USB enumeration issues with verions after v1_5_1b7

Tjeerdie, 2023-04-16:
Using FreeJoy_v1_5_1b7 there is no issue with the usb enumeration. If i install any version after v1_5_1b7 the USB enumeration fails. It is possible that i am using a fake STM32F103 i am not sure, but if it works for older versions of the software what would be the cause of issues with newer versions?

kozinalexey, 2023-04-19:
I have same problem when use BIN file for flash using serial adapter and flash loader. Try to use HEX fle

Tjeerdie, 2023-04-21:
II fixed the issue:

It seems the manual for serial upload is out of date. Or the binaries in the releases are not complete (app + bootloader together in one *.bin. The application *.bin in the releases does not include the bootloader. But it needs the bootloader to start at the correct entry point. So uploading the application *.bin only does not work.

You first need to upload the FreeJoy bootloader. This one is not published so i build it using  the repo and uploaded the *.bin. After that the device is recognized via usb in boot loader mode. 

When it is in bootloader mode you can upload the application *.bin using the freeJoyQT application.

closed 2023-04-21 by Tjeerdie