FreeJoy-Team/FreeJoy #208: Adding UART output

relaxibus, 2023-05-27:
Hi, I found the Freejoy project, very impressive. I would like to use it for a RC TX project, but instead of using the USB I would need a UART output to send control commands. I designed a UART protocol that reflect the USB descriptor variables such as x/y/z/xr/etc. Would it possible that you add the UART output option?

relaxibus, 2023-06-15:
Not really, the idea is to use it to control RC models (planes, copters, boats and cars) without any OS in the middle. The serial output (SERaero open source protocol) will passed to an ESP32 (or any other MCU) to convert the SERaero into PP, SBUS, CRSF, ELRS, MAVLink or CAN. Here a PoC using a USB-HID to SERaero converter for any USB gaming device. This could be used, but it it would be a more reliable way to avoid the USB-HID to SERaero block and export SERaero direct out from the Freejoy board: https://www.youtube.com/watch?v=L_21r6xz2tw&t=49s

relaxibus, 2023-07-01:
@MMahyaRA do you see a way to add this to Feejoy?

meghang-101, 2024-03-14:
> Not really, the idea is to use it to control RC models (planes, copters, boats and cars) without any OS in the middle. The serial output (SERaero open source protocol) will passed to an ESP32 (or any other MCU) to convert the SERaero into PP, SBUS, CRSF, ELRS, MAVLink or CAN. Here a PoC using a USB-HID to SERaero converter for any USB gaming device. This could be used, but it it would be a more reliable way to avoid the USB-HID to SERaero block and export SERaero direct out from the Freejoy board: https://www.youtube.com/watch?v=L_21r6xz2tw&t=49s
> 
> Here the SERaero protocol specs: [SERaero_C2-16B_spec_v1.1.pdf](https://github.com/FreeJoy-Team/FreeJoy/files/11755241/SERaero_C2-16B_spec_v1.1.pdf)

You need not use FreeJoy for this application. FreeJoy is purposely made for HID interface between STM32 chips and your desktop. For your operation, you can directly create your own driver. Just put the analog data to a UART buffer and transmit it directly. Or if you want to use the SERaero, directly boot the chip with the SERaero driver. No need of a HID driver.