FreeJoy-Team/FreeJoy #209: Question:

MMahyaRA, 2023-06-13:
How can I add more sliders? For example, currently, we have 8 analog inputs, but I want to have 10 or 12.

wizhippo, 2023-06-13:
If I recall windows does not play well with more. For example ref https://github.com/NicoHood/HID/issues/8#issuecomment-143512589

You would have to have make a multi joystick device descriptor to get past the windows limitation and act as two joysticks, which would require both firmware and configuration app changes to support.

The inexpensive cost of another controller may not justify the effort to implement.  Though it would be a nice to have.

MMahyaRA, 2023-06-13:
> If I recall windows does not play well with more. For example ref [NicoHood/HID#8 (comment)](https://github.com/NicoHood/HID/issues/8#issuecomment-143512589)
> 
> You would have to have make a multi joystick device descriptor to get past the windows limitation and act as two joysticks, which would require both firmware and configuration app changes to support.
> 
> The inexpensive cost of another controller may not justify the effort to implement. Though it would be a nice to have.

Can I achieve this with an stm32f103c8 and the freeJoy library? I mean, can I have multiple gamepads and remove the axis limitations?

wizhippo, 2023-06-13:
No, you would need additional stm32f103c8 as it stands today without modifying the firmware and the configurator.

MMahyaRA, 2023-06-13:
> No, you would need additional stm32f103c8 as it stands today without modifying the firmware and the configurator.

In this case, can I connect both of them to a single USB and consider different VIDs for each one?

wizhippo, 2023-06-13:
They would need to be 2 separate USB devices.

Yes I would bump the VID on one of them so your computer can distinguish them.  I so this with a throttle quadrant and custom yoke I use.

MMahyaRA, 2023-06-13:
> They would need to be 2 separate USB devices.
> 
> Yes I would bump the VID on one of them so your computer can distinguish them. I do this with a throttle quadrant and custom yoke I use.

Another question, can I write my own program for stm32f103c8 regardless of the software provided for its configuration?

wizhippo, 2023-06-13:
If you have the skill, an programmer like the st-link and access to the debug pins. no reason why you couldn't.

You can treat as an arduino and leverage the arduino ecosystems if you wanted,

closed 2023-06-13 by MMahyaRA

LUCKandII, 2024-03-03:
> No, you would need additional stm32f103c8 as it stands today without modifying the firmware and the configurator.

Hi, if I have multiple freejoy devices(one for flight stick and one for throttle, two separate STM32F103C8T6 boards), can they both work without modifying the firmware?