FreeJoy-Team/FreeJoy #211: freejoy and TH cougar  grip

teeshq, 2023-08-01:
Hi i have some issue with working with cougar grip(if i connect it  to virpil WARBRD its work so im sure shift register in grip is ok).
after connecting grip some buttons work OK, some not at all, and other have multiple press(example pressing trim button up cause 3 physical buttons light up ). 
I also to some measurement and all data pins on warbrd give higher voltage like 0.8v, 3.6v, 2.6v . On freejoy pins assigned to latch data clk , give much lover value cant this make this behavior ??
some photos below also one short video.

thanks a lot

[image]
[image]
[image]
[image]
[https://cdn.discordapp.com/attachments/1135602674973294622/1135650174258139177/VID_20230731_200651.mp4](url)

yzhanping, 2023-08-03:
I think your problem is on the expansion board. My expansion board can normally replace the Tumaster a10c /F18C grip, recognize 24 triggers; and can realize 128 triggers in series;
They are like this
[image]
[image]

teeshq, 2023-08-03:
HI thanks yes could be original thrustmaster shift reg board have some different logic (but still it use 4021 IC) and grip work ok with virpil base :/. 
probably i will just get my own pcb with HC165
[image]

vostrenkov, 2023-08-04:
Arduino (and I guess warbrd also) uses 5V logic but STM32 has 3.3V logic
CD4021 may have some issues with 3.3V logic when connected over long wires or having high connections capacity.

Using voltage level translator most likely would solve the isuue

teeshq, 2023-08-04:
> Arduino (and I guess warbrd also) uses 5V logic but STM32 has 3.3V logic CD4021 may have some issues with 3.3V logic when connected over long wires or having high connections capacity.
> 
> Using voltage level translator most likely would solve the issue

Hi thanks can u advise what translator to use im quite noob you mean somethings like that https://makershop.ie/CJMCU-0102 .
or https://ie.farnell.com/texas-instruments/cd40109be/ic-level-shifter-quad-16dip/dp/3120902

ghawkgu, 2023-08-06:
@teeshq 
Regarding the issue that multiple buttons being triggered. I had similar issue when I tried to wire up the cougar throttle.
#179 

However, the throttle grip uses a button matrix instead of a shift register.
I doubt there's some defect in the logic of reading physical button status.

TomHVR, 2024-01-16:
see this firmware in my issue, it might solve your problem (and the change hopefully will be folded into a new version of freejoy)

https://github.com/FreeJoy-Team/FreeJoy/issues/222