FreeJoy-Team/FreeJoy #212: STM32 FreeJoy lightweight library

MMahyaRA, 2023-08-03:
Hello, I want to create a set of controllers. These controllers each have a separate stm32 that reads their inputs, processes the data, and then sends it to the main microcontroller f103c8. My goal is to connect it to the computer using FreeJoy. However, I'm not sure how to accomplish this. It would be great if FreeJoy could be a library for stm32, just like what we have for Arduino (joystick library).

Do you have any better suggestions?

vostrenkov, 2023-08-04:
Hi, you can use freejoy code without USB part
Default config is defined in main.h file 

The only addition that needed is the output of the results to some communication interface (e.g. UART)

MMahyaRA, 2023-08-04:
> Hi, you can use freejoy code without USB part Default config is defined in main.h file
> 
> The only addition that needed is the output of the results to some communication interface (e.g. UART)

It's quite strange that the build option in the IDE is disabled; however, I have only used buttons and axes. The first microcontroller reads the status of the buttons and then sends it to the main microcontroller through serial communication. The second microcontroller also reads the values of the axes and sends them to the main microcontroller in a similar way. Now, I want the main microcontroller to combine these values and send them to the computer via USB. I have two problems: firstly, I don't know exactly where in the FreeJoy program I should send them to the USB, and secondly, the FreeJoy program is too heavy for my application. I want to remove some of the unused headers and codes to make the code smaller. Can you help me more?