FreeJoy-Team/FreeJoy #214: USB not Recognized

kapoorshivam100, 2023-08-11:
After flashing the firmware in blue pill, I try to connect it via USB which is not recognized by the computer. Kindly Help me out

mupfdev, 2023-08-31:
This is a very common problem. Could you check the pull-up resistors on the USB data lines? If you don't have the proper equipment, the marking would help.

It should be 1k5, however, some clones use 4k7 or 10k resistors.

Best regards,
Michael

mupfdev, 2023-08-31:
Also try to erase the chip prior to flashing it. This helped me once.

al-ship, 2023-11-20:
If flashing through usb-uart bin file it fails to recognize. You need bootloader.bin, wich not included in release. You can build it from sources and flash. Doc is a bit outdated

al-ship, 2023-11-21:
OK, thank you! But [wiki](https://github.com/FreeJoy-Team/FreeJoyWiki/blob/master/eng/Linux-Guide.md) says that "Make sure to use the .bin version."

vostrenkov, 2023-11-21:
Yeah, there is a mistake in Linux guide
It is not frequently used and may be a bit outdated
Fixed now

trasz, 2024-03-12:
Just got bitten by this, and can confirm: if you're trying to flash FreeJoy into a fresh Blue Pill, you need to use:

st-flash --format ihex write FreeJoy_v1_7_1b3.hex

Using FreeJoy_v1_7_jb3.bin instead of hex resulted in BluePill that wouldn't get enumerated when connected to USB.

butasi, 2024-11-10:
Is it possible to get this running on non blue pill board? Like STM32F107RCT6