FreeJoy-Team/FreeJoy #222: CD4021 fix in main version

RightRudderLeftStick, 2024-01-07:
I know Magners made this firmware for me ages ago to get my TM grips compatibility working:

[FreeJoy_1_7_1b4_jumpnshootman_v3.zip](https://github.com/FreeJoy-Team/FreeJoy/files/13854641/FreeJoy_1_7_1b4_jumpnshootman_v3.zip)

I believe it had something to do with the timing. Wonder if this will get merged into the release version in the future.

vostrenkov, 2024-01-11:
note for the future: clock frequency was reduced there

jochenschmich, 2025-04-24:
@RightRudderLeftStick / @vostrenkov Hi there - out of interest: how exactly was the code changed?

vostrenkov, 2025-04-24:
> [@RightRudderLeftStick](https://github.com/RightRudderLeftStick) / [@vostrenkov](https://github.com/vostrenkov) Hi there - out of interest: how exactly was the code changed?

**SHIFTREG_TICK_DELAY** value was increased:

```c
/**
  ******************************************************************************
  * @file           : shift_registers.h
  * @brief          : Header for shift_registers.c file.
  ******************************************************************************
  */

/* Define to prevent recursive inclusion -------------------------------------*/
#ifndef __SHIFT_REGISTERS_H__
#define __SHIFT_REGISTERS_H__

#include "common_types.h"
#include "periphery.h"

#define SHIFTREG_TICK_DELAY			1

void ShiftRegistersInit(dev_config_t * p_dev_config);
void ShiftRegistersGet (uint8_t * raw_button_data_buf, dev_config_t * p_dev_config, uint8_t * pos);

#endif 	/* __SHIFT_REGISTERS_H__ */
```

jochenschmich, 2025-04-24:
@vostrenkov Thanks for the super-quick answer. May I ask to what specific value? It has been set to `1` with this commit: https://github.com/FreeJoy-Team/FreeJoy/commit/ddc1301e53ebae1df5394a5eee52faf83a270934

Has been set to `5` earlier.

Has anybody set this to another value to improve support for a Thrustmaster Cougar Handle connected via a long cable?

vostrenkov, 2025-04-24:
No chance I remember which exact value that was, but I assume it is something in between 1 and 5

TomHVR, 2025-04-24:
if there is a future update, is this something that could be configurable with the UI?

vostrenkov, 2025-04-25:
> if there is a future update, is this something that could be configurable with the UI?

I'd say that moving to another platform from bluepill is the main goal for the possible update
The bluepill version of firmware most likely stays as is