FreeJoy-Team/FreeJoy #225: NotAnIssue - Merging of opposite axis

DukeOfDream, 2024-01-11:
Hi, i'm kinda new to GitHub and don't know if there's a request section... Is it possible to add an option to merge axis? I'm working on a fried Logitech X-52 pro joystick that i want to convert to free-joy but it uses 2 opposing hall sensors for each of the 2 axis. Basically 2 static hall sensors in the middle glued together and 2 moving magnets... While one is raising the value, the other is dropping it... Pretty much one hall sensor deals with half the axis while the other deals with the other half...  I'm stuck at it for about a month and can't figure a way to make it work... I could use just one of the 2 hall sensors, but then the resolution on one half of the axis becomes really poor..... Can you add this functionality to the project? Thanks in advance and thank you to everyone that has worked on this project to make our lives a LOT easier <3

vostrenkov, 2024-01-11:
Hi,
I'm not sure if I understand your case correctly, but aren't the Addition/Subtraction functions for axis what you looking for?

vostrenkov, 2024-01-11:
See 'Combined Axes'
https://github.com/FreeJoy-Team/FreeJoyWiki/blob/master/eng%2FAxis-configuration.md

DukeOfDream, 2024-01-12:
> Hi, I'm not sure if I understand your case correctly, but aren't the Addition/Subtraction functions for axis what you looking for?

> See 'Combined Axes' https://github.com/FreeJoy-Team/FreeJoyWiki/blob/master/eng%2FAxis-configuration.md

You guys are right... I've read the "Combined Axes" section before but i didn't understand it properly i guess... I thought it could merge 2 logical axes from a single physical axes... I was expecting to see a physical axis from the "function axis" drop down menu, instead it mentions logical axis (X-Y-Z etc) What i was looking for was the minus function, seems to be working correctly... i'll be testing it out thoroughly :)

closed 2024-01-12 by DukeOfDream