FreeJoy-Team/FreeJoy #231: file size is bigger than the flash memory size.!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

ikuunx, 2024-04-06:
when I start to enter the program, the st-link V2 app said"file size is bigger than the flash memory size!!!!

who can help me solve this problem?

19:50:32 : ST-LINK SN : B55B5A1A0000000039CFEF01
19:50:32 : V2J34S7
19:50:32 : Connected via SWD.
19:50:32 : SWD Frequency = 4,0 MHz.
19:50:32 : Connection mode : Normal.
19:50:32 : Debug in Low Power mode enabled.
19:50:32 : Device ID:0x412 
19:50:32 : Device flash Size : 32KBytes
19:50:32 : Device family :STM32F10xx Low-density
19:50:33 : Flash memory erased.
19:52:07 : ST-LINK SN : B55B
19:52:07 : V2J34S7
19:52:07 : Connected via SWD.
19:52:07 : SWD Frequency = 4,0 MHz.
19:52:07 : Connection mode : Normal.
19:52:07 : Debug in Low Power mode enabled.
19:52:07 : Device ID:0x412 
19:52:07 : Device flash Size : 32KBytes
19:52:07 : Device family :STM32F10xx Low-density
19:52:21 : [FreeJoy_v1_7_1b2.hex] opened successfully.
19:52:21 : [FreeJoy_v1_7_1b2.hex] checksum : 0x0096951C 
19:53:10 : File size is bigger than the flash memory size.
19:53:37 : File size is bigger than the flash memory size.

void269, 2024-08-21:
@ikuunx you likely have the same issue as I had.  your Blue Pill has 32kb of flash but you need the 64kb version...
https://github.com/FreeJoy-Team/FreeJoy/issues/241

void269, 2024-08-21:
I have also asked to have the minimum flash memory be stated in the Wiki page https://github.com/FreeJoy-Team/FreeJoyWiki/issues/81

UID163844189, 2025-03-12:
You bought STM32F103C6T6 which has 32KB flash. But you need STM32F103C(8|B)T6 which has 64 or 128 KB flash. Their codes and firmwares are compatible (except for your firmware larger than 64 KB). All these STM32s can be used as Bluepill.