FreeJoy-Team/FreeJoy #237: Disconnects after receiving 1000-1500 packets (acc. to debug), PC13 LED also dims after a second.

amateurstock, 2024-07-12:
It struggles to stay connected with the FreeJoy configurator before disconnecting. I've connected it via microUSB. Pressing the reset button does reconnect it, but the above issue still persists. Is this just a bad USB connection or a fake board?

[image]

amateurstock, 2024-07-13:
Annnnd it turns out my board is indeed fake. I turned on flasher mode, and the connection between the board and the interface was secure regardless of how i wiggled the USB cable. I've just answered my own question.

void269, 2024-08-23:
@amateurstock How do you know the board is fake?  Mine is doing the same thing and I tested it with ST's genuine app which said it was not fake.

amateurstock, 2024-08-25:
Frankly, I don't really have a good answer. Ever since my luck ran out with this repo I just decided I code my own project that will work with my device (and hopefully the code is generic enough that it might also work with other fake boards).

The bluepill dev board is not really ST's own product, and you're likely to find out-of-spec dev boards if you try to cheap out with online shops like Alibaba (or Shopee where I'm from). The chip is ST's, but the board might deviate a lot from the original spec. 

The main takeaway is code you find for the bluepill may or may not work with your bluepill because of what I've said above. My only regret is not choosing to save up and buy the more expensive boards like the NUCLEOs and the DISCOs; plenty of projects are already built upon them and they're really neat (built-in serial to USB, better peripherals and performance, etc.).

closed 2024-08-25 by amateurstock