FreeJoy-Team/FreeJoy #244: Blue Pill Can't be detected by FreeJoy Configurator

Idahoooo, 2024-12-07:
Use STM32 ST-LINK Utility to flash firmware(FreeJoy_v1_7_1b3)successfully , connect it to the computer by a ST-LINK V2,but FreeJoy configurator(v1.7.1b5) still shows " disconnected".

foxhunter-73, 2025-01-11:
Hello !! I think tha same prb.
I bought 2 BluePill (original chip st), first erase, after Flashing... by st-link,  all ok!
In the configurator app it's connect for 2 second, after disconnect... reconnect... and disconnect
I was unable to understand why... i'm frustated  :-P

luca-mosca, 2025-02-04:
Same here, please, any solutions?

UID163844189, 2025-03-12:
Hi everyone, 
I found similar problem with firmware 1.7.1b3. And I flashed the 1.7.1b2 firmware, with no problem on using until now from Feb 28, using config app ver v1.7.1b5. 
Can you try the 1.7.1b2 version of firmware and see if the problems still?

Edermg05, 2025-04-24:
> i'll try when i can. At the moment i solved by programming and arduino avr family, and do it all by programming. Thank 4 you'r answer

Hi, I am in the same situation. Could you develop a little bit further on how did you manage to program the firmware in Arduino? I would like to try it before buying another blue pill

Edermg05, 2025-04-24:
> No i program direct it using the ide and the joystick library. Did you try tu use the the 1.7.1b2 version of firmware?

Yes, I tried from 1b0 to 1b5 and had the same issues with all of them; Connected for a couple of seconds, disconnected, reconnected for a short bit again and permanently disconnected until I manually press the reset switch.

Do you, by any chance, still have the arduino code or any kind of guidance on how to do it? By what I understand, with the arduino code you wrote, you can't use FreeJoy, right? Are you using a self programmed code independent of it?

foxhunter-73, 2025-04-24:
No, there was another tool like freejoy 4 arduino but wasn't close.
I programm in c and i use some libraries.
I print a stick and throttle to play flight sym but i've some experience in electronics.... What do you wanna do?

foxhunter-73, 2025-04-24:
check this libraries
https://github.com/vsulako/TLE5010
it's no so difficult!
There are an interesting 'world' about gmr

Edermg05, 2025-04-24:
> check this libraries https://github.com/vsulako/TLE5010 it's no so difficult! There are an interesting 'world' about gmr

Well, that library is for radial movements, not linear, but that might be something to begin with. I might take a deeper look at Arduino Libraries. Thanks.

UID163844189, 2025-06-16:
I noticed that you must flash the .hex file into a new stm32. .bin is only used for update as I know. 
And, according to Windows's some bugs, change to another usb port may works. I don't know if reboot computer also works in this condition.