FreeJoy-Team/FreeJoy #245: Add ForceFeedback pls.

nhy6iii, 2024-12-17:
I would love to see ability to steer force feedback motor by PWM signal.
received PWM signal steer Torque output.
Is a any chance for another use of PWM signals?

wizhippo, 2024-12-17:
Are you looking for native USB FFB support? It's not a trivial item to add.  How would you envision the configuration would be done in the config UI?

Here is a related project that may help you find a way to do what you want.
https://github.com/YukMingLaw/ArduinoJoystickWithFFBLibrary

nhy6iii, 2024-12-18:
If I understand correctly from Windows perspective, it's the same as switching LED on the controller. difference is a value isn't 1-0 but from some range like -1000 to +1000.
and FreeJoy would convert that value to PWM signal ex. -1000 is 10% filll, 0 is a range from 49-51% fill but sending 50% only, +1000 is 90%fill.
in UI during Axis configuration would be another option "Add ForceFeedback" and chose on with pin is PWM output for that.

motor drive, safety, limits, that would be another microcontroller
And requesting specific force/torque would game directly or program like xpForce.

nhy6iii, 2024-12-31:
found solution:
https://github.com/Ultrawipf/OpenFFBoard

closed 2025-01-06 by nhy6iii