FreeJoy-Team/FreeJoy #247: Freejoy questions

harrinator-yxz, 2025-02-18:
Hey, I am building a diy 3d printed f16 flight stick. Originally I was looking to do the electronics via Bluetooth with an esp32, then I found this and I have some questions

1. Will an mpu6050 work with freejoy, I am using potentiometers for x and y but for z I was thinking of using a gyro, does freejoy support gyros and have gyro filtering or anything like that.
2. The blue pill, I am looking to buy most of my parts from sites like aliexpress and temu, is this a blue pill that will work: https://share.temu.com/zisGNzqLJuA
3. In terms of wiring I can just wire up all my parts to the blue pill eg buttons one pin to ground and the other to a pin on blue pill and that will work?. 
4. Does this kind of emulate an Xbox controller or something like that?

Hope these are relevant questions and I can have some feedback :)

UID163844189, 2025-03-12:
2. Yes, but beware of fake STM32s. Fake chips may have compability issue, stability issue, or other unexpected issues. In China, a 64 KB flash Bluepill should cost around 10 CNY, a piece of STM32F103C8T6 should cost 4~5 CNY depends on global market. 
3. Yes, it works, but we used to connect one side to VCC, another side connect to GPIO. but both ways are available to this project, as I tested. 
4. Functionally yes. In games like X-Plane, ETS2 these games needing specified controllers (yoke, steering wheel) and a lot of swiches, buttons, ones you finished setup the Freejoy, you can go into the contoller configurating page in game, it will be recoginized, with other selling controllers pararelly like Logitech G923 this. Then you can assign every button, or switch to a function in the plane or truck, such as throttle, flaps, reverses (in flight simulator), or lights, engine start/stop, parking brakes (in ETS2), just like the button on your bought controller. 
5. Computer will just recoginized it as a controller. No more software or batches need to use it (except the configurating app and it is unnecessary to run in your normal gaming). 
6. Thus, Arduino Uno do not have a native USB port. STM32F1 does, which means it can be configurate to "be recoginized as a joystick by computer". Uno can't, because the USB port on it can only be recoginized as a serial port (show "COM Port" in OS). Then an alternative software is needed to convert the serial data to a "joystick-as" data. 

English is not my first language and there may be grammar error, spelling error in my text. Please do not pay attention on it.

harrinator-yxz, 2025-03-13:
my temu cs32 workd

works fine - great

closed 2025-03-13 by harrinator-yxz